siemens/meta-iot2050 #481: MiniPCI and USB Dongle Issue

ronaldhernandezm, 2023-09-16:
Hello, i was recently trying to use a miniPCI module: SIERRA MC7700 and a USB Dongle AWUS036ACS - ALFA
I have the exampleimage 1.3.1
The problem with it is that even with an SIM card and the module connected the IOT detects it as USB :
[image]
And i cannot do nothing to use it ... ,
[image]
[image]
That is what i have right now...

In adittion , i have some weird issue when i try to install the drivers for the USB dongle , i receive a meesage that says that i have an wrong kernel header, but i was trying everything and nothing works :/ .
Im using this to guide me: https://github.com/morrownr/8821au-20210708

I would really appreciate your help with this, thanks in advance!!

jan-kiszka, 2023-09-16:
Look like our kernel config is missing `CONFIG_USB_SERIAL_SIERRAWIRELESS`. You could add that to `recipes-kernel/linux/files/iot2050_defconfig_extra.cfg` and rebuild your own image.

ronaldhernandezm, 2023-09-16:
Hello , thanks for the prompt response
That CONFIG_USB_SERIAL_SIERRAWIRELESS is a filé or a code line to write on that address? Sorry i am new on this

jan-kiszka, 2023-09-16:
You will have to add a line with `CONFIG_USB_SERIAL_SIERRAWIRELESS=m` to that `iot2050_defconfig_extra.cfg`.

PR #483 (2023-09-20): linux-iot2050: Enable support for Sierra Wireless USB modem
This should resolve #481, namely adding support for the MC7700.