siemens/meta-iot2050 #485: Building fails on tag V01.02.02

Erokos, 2023-09-21:
Hi,

we checked out the tag `V01.02.02` and initiated the build using the following command:

`./kas-container build kas-iot2050-example.yml:kas/opt/package-lock.yml`

contrary to expectations the build fails with the following error:

```
ERROR: openssl-latest-r0 do_dpkg_build: Error executing a python function in exec_python_func() autogenerated:

The stack trace of python calls that resulted in this exception/failure was:
File: 'exec_python_func() autogenerated', lineno: 2, function: <module>
     0001:
 *** 0002:do_dpkg_build(d)
     0003:
File: '/work/isar/meta/classes/dpkg-base.bbclass', lineno: 184, function: do_dpkg_build
     0180:    lock = bb.utils.lockfile(d.getVar("REPO_ISAR_DIR") + "/isar.lock",
     0181:                             shared=True)
     0182:    bb.build.exec_func("dpkg_do_mounts", d)
     0183:    try:
 *** 0184:        bb.build.exec_func("dpkg_runbuild", d)
     0185:    finally:
     0186:        bb.build.exec_func("dpkg_undo_mounts", d)
     0187:        bb.utils.unlockfile(lock)
     0188:}
File: '/work/isar/bitbake/lib/bb/build.py', lineno: 251, function: exec_func
     0247:    with bb.utils.fileslocked(lockfiles):
     0248:        if ispython:
     0249:            exec_func_python(func, d, runfile, cwd=adir)
     0250:        else:
 *** 0251:            exec_func_shell(func, d, runfile, cwd=adir)
     0252:
     0253:    try:
     0254:        curcwd = os.getcwd()
     0255:    except:
File: '/work/isar/bitbake/lib/bb/build.py', lineno: 452, function: exec_func_shell
     0448:    with open(fifopath, 'r+b', buffering=0) as fifo:
     0449:        try:
     0450:            bb.debug(2, "Executing shell function %s" % func)
     0451:            with open(os.devnull, 'r+') as stdin, logfile:
 *** 0452:                bb.process.run(cmd, shell=False, stdin=stdin, log=logfile, extrafiles=[(fifo,readfifo)])
     0453:        finally:
     0454:            os.unlink(fifopath)
     0455:
     0456:    bb.debug(2, "Shell function %s finished" % func)
File: '/work/isar/bitbake/lib/bb/process.py', lineno: 182, function: run
     0178:        if not stderr is None:
     0179:            stderr = stderr.decode("utf-8")
     0180:
     0181:    if pipe.returncode != 0:
 *** 0182:        raise ExecutionError(cmd, pipe.returncode, stdout, stderr)
     0183:    return stdout, stderr
Exception: bb.process.ExecutionError: Execution of '/build/tmp/work/iot2050-debian-arm64/openssl/latest-r0/temp/run.dpkg_runbuild.126684' failed with exit code 2:
dpkg-buildpackage: info: source package openssl
dpkg-buildpackage: info: source version 1.1.1n-0+deb11u4+iot2050
dpkg-buildpackage: info: source distribution UNRELEASED
dpkg-buildpackage: info: source changed by Unknown maintainer <unknown@example.com>
dpkg-architecture: warning: specified GNU system type aarch64-linux-gnu does not match CC system type x86_64-linux-gnu, try setting a correct CC environment variable
 dpkg-source -I --before-build .
dpkg-buildpackage: info: host architecture arm64
 debian/rules clean
dh clean --without autoreconf
   debian/rules override_dh_auto_clean
make[1]: Entering directory '/home/builder/openssl/openssl-latest'
rm -rf build_static build_shared
test -z "" || for opt in ; \
do \
        set -xe; \
        rm -rf build_$opt; \
[... 210 lines cut ...]
 openssl-latest/test/recipes/80-test_ssl_old.t
 openssl-latest/test/rsa_test.c
 openssl-latest/test/sm2_internal_test.c
 openssl-latest/test/sslapitest.c
 openssl-latest/test/ssltest_old.c
 openssl-latest/test/testutil/format_output.c
 openssl-latest/test/testutil/tests.c
 openssl-latest/test/verify_extra_test.c
 openssl-latest/test/x509_time_test.c
 openssl-latest/util/mkdir-p.pl
dpkg-source: error: aborting due to unexpected upstream changes, see /tmp/openssl_1.1.1n-0+deb11u4+iot2050.diff.MdLX1a
dpkg-source: info: you can integrate the local changes with dpkg-source --commit
dpkg-buildpackage: error: dpkg-source -I -b . subprocess returned exit status 2
WARNING: exit code 2 from a shell command.

ERROR: Logfile of failure stored in: /build/tmp/work/iot2050-debian-arm64/openssl/latest-r0/temp/log.do_dpkg_build.126684
ERROR: Task (/repo/recipes-security/openssl/openssl_latest.bb:do_dpkg_build) failed with exit code '1'
NOTE: Tasks Summary: Attempted 285 tasks of which 84 didn't need to be rerun and 1 failed.

Summary: 1 task failed:
  /repo/recipes-security/openssl/openssl_latest.bb:do_dpkg_build
Summary: There was 1 ERROR message shown, returning a non-zero exit code.
2023-09-21 08:53:51 - ERROR    - Command returned non-zero exit status 1
```
Can you please help with this issue? Tags should point to something reproducible and stable.

jan-kiszka, 2023-09-21:
Please try building head of the stable/V01.02 branch instead.

Erokos, 2023-09-21:
Thank you for the prompt response, on this branch I get this:

```
ERROR: openssl-latest-r0 do_apt_unpack: Execution of '/build/tmp/work/iot2050-debian-arm64/openssl/latest-r0/temp/run.do_apt_unpack.41089' failed with exit code 1:
cp: -r not specified; omitting directory '/downloads/deb-src/iot2050-debian/openssl/bullseye'
WARNING: exit code 1 from a shell command.

ERROR: Logfile of failure stored in: /build/tmp/work/iot2050-debian-arm64/openssl/latest-r0/temp/log.do_apt_unpack.41089
Log data follows:
| DEBUG: Executing shell function do_apt_unpack
| cp: -r not specified; omitting directory '/downloads/deb-src/iot2050-debian/openssl/bullseye'
| WARNING: exit code 1 from a shell command.
| ERROR: Execution of '/build/tmp/work/iot2050-debian-arm64/openssl/latest-r0/temp/run.do_apt_unpack.41089' failed with exit code 1:
| cp: -r not specified; omitting directory '/downloads/deb-src/iot2050-debian/openssl/bullseye'
| WARNING: exit code 1 from a shell command.
|
ERROR: Task (/repo/recipes-security/openssl/openssl_latest.bb:do_apt_unpack) failed with exit code '1'
NOTE: Tasks Summary: Attempted 217 tasks of which 0 didn't need to be rerun and 1 failed.

Summary: 1 task failed:
  /repo/recipes-security/openssl/openssl_latest.bb:do_apt_unpack
Summary: There was 1 ERROR message shown, returning a non-zero exit code.
2023-09-21 11:50:36 - ERROR    - Command returned non-zero exit status 1
```

Erokos, 2023-09-21:
I've also tried commenting out the `DEPENDS += openssl` line and removed the node-red packages from the recipe because it was then failing because of them. Now I got the farthest but still fails:
```
Errors were encountered while processing:
 customizations-example
E: Sub-process /usr/bin/dpkg returned an error code (1)
WARNING: exit code 100 from a shell command.

ERROR: Logfile of failure stored in: /build/tmp/work/iot2050-debian-arm64/iot2050-image-example-iot2050-wic-img/1.0-r0/temp/log.do_rootfs_install.673
ERROR: Task (/repo/recipes-core/images/iot2050-image-example.bb:do_rootfs_install) failed with exit code '1'
NOTE: Tasks Summary: Attempted 269 tasks of which 255 didn't need to be rerun and 1 failed.

Summary: 1 task failed:
  /repo/recipes-core/images/iot2050-image-example.bb:do_rootfs_install
```
I wonder is this possible to build?

jan-kiszka, 2023-09-21:
Was this a clean rebuild attempt? I will retry locally as well.

Erokos, 2023-09-25:
Yes it was a clean rebuild. Have you managed to successfully build it?

jan-kiszka, 2023-09-25:
I did a clean build locally with your command line and 3c0f8553fe2ec15c5f241d988b06699d20b7d5bd (stable/V01.02) without problems.

closed 2023-10-03 by jan-kiszka