siemens/meta-iot2050 #486: disk  full new build

maxvaneck, 2023-09-22:
greetings i just build a new image from master   with only example image turned on . so far so good. the problem is that   disk is 100% full on a new install .  included below  are some commands and output 
df -h
Filesystem              Size  Used Avail Use% Mounted on
udev                    449M     0  449M   0% /dev
tmpfs                   500M  912K  500M   1% /run
/dev/mmcblk0p4          723M  723M     0 100% /
/dev/disk/by-label/var  638M  341M  251M  58% /var
overlay                 638M  341M  251M  58% /etc
tmpfs                   465M     0  465M   0% /dev/shm
tmpfs                   5.0M     0  5.0M   0% /run/lock
tmpfs                   465M     0  465M   0% /tmp
/dev/mmcblk0p7           21G  512K   20G   1% /home
tmpfs                    93M     0   93M   0% /run/user/0
lsblk
NAME        MAJ:MIN RM   SIZE RO TYPE MOUNTPOINTS
mtdblock0    31:0    0   1.5M  0 disk 
mtdblock1    31:1    0     2M  0 disk 
mtdblock2    31:2    0     3M  0 disk 
mtdblock3    31:3    0   128K  0 disk 
mtdblock4    31:4    0   128K  0 disk 
mtdblock5    31:5    0    64K  0 disk 
mtdblock6    31:6    0   7.7M  0 disk 
mtdblock7    31:7    0   1.5M  0 disk 
mmcblk0     179:0    0  29.5G  0 disk 
|-mmcblk0p1 179:1    0  16.1M  0 part 
|-mmcblk0p2 179:2    0  41.7M  0 part 
|-mmcblk0p3 179:3    0  41.7M  0 part 
|-mmcblk0p4 179:4    0     4G  0 part /
|-mmcblk0p5 179:5    0     4G  0 part 
|-mmcblk0p6 179:6    0 665.6M  0 part /var
`-mmcblk0p7 179:7    0  20.8G  0 part /home
fdisk -l
Disk /dev/ram0: 4 MiB, 4194304 bytes, 8192 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 4096 bytes
I/O size (minimum/optimal): 4096 bytes / 4096 bytes

Disk /dev/ram1: 4 MiB, 4194304 bytes, 8192 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 4096 bytes
I/O size (minimum/optimal): 4096 bytes / 4096 bytes

Disk /dev/ram2: 4 MiB, 4194304 bytes, 8192 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 4096 bytes
I/O size (minimum/optimal): 4096 bytes / 4096 bytes

Disk /dev/ram3: 4 MiB, 4194304 bytes, 8192 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 4096 bytes
I/O size (minimum/optimal): 4096 bytes / 4096 bytes

Disk /dev/ram4: 4 MiB, 4194304 bytes, 8192 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 4096 bytes
I/O size (minimum/optimal): 4096 bytes / 4096 bytes

Disk /dev/ram5: 4 MiB, 4194304 bytes, 8192 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 4096 bytes
I/O size (minimum/optimal): 4096 bytes / 4096 bytes

Disk /dev/ram6: 4 MiB, 4194304 bytes, 8192 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 4096 bytes
I/O size (minimum/optimal): 4096 bytes / 4096 bytes

Disk /dev/ram7: 4 MiB, 4194304 bytes, 8192 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 4096 bytes
I/O size (minimum/optimal): 4096 bytes / 4096 bytes

Disk /dev/ram8: 4 MiB, 4194304 bytes, 8192 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 4096 bytes
I/O size (minimum/optimal): 4096 bytes / 4096 bytes

Disk /dev/ram9: 4 MiB, 4194304 bytes, 8192 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 4096 bytes
I/O size (minimum/optimal): 4096 bytes / 4096 bytes

Disk /dev/ram10: 4 MiB, 4194304 bytes, 8192 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 4096 bytes
I/O size (minimum/optimal): 4096 bytes / 4096 bytes

Disk /dev/ram11: 4 MiB, 4194304 bytes, 8192 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 4096 bytes
I/O size (minimum/optimal): 4096 bytes / 4096 bytes

Disk /dev/ram12: 4 MiB, 4194304 bytes, 8192 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 4096 bytes
I/O size (minimum/optimal): 4096 bytes / 4096 bytes

Disk /dev/ram13: 4 MiB, 4194304 bytes, 8192 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 4096 bytes
I/O size (minimum/optimal): 4096 bytes / 4096 bytes

Disk /dev/ram14: 4 MiB, 4194304 bytes, 8192 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 4096 bytes
I/O size (minimum/optimal): 4096 bytes / 4096 bytes

Disk /dev/ram15: 4 MiB, 4194304 bytes, 8192 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 4096 bytes
I/O size (minimum/optimal): 4096 bytes / 4096 bytes

Disk /dev/mmcblk0: 29.54 GiB, 31719424000 bytes, 61952000 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 512 bytes / 512 bytes
Disklabel type: gpt
Disk identifier: EA4219F7-DCE8-49B2-A227-EBE7CA314B8C

Device            Start      End  Sectors   Size Type
/dev/mmcblk0p1     2048    34975    32928  16.1M EFI System
/dev/mmcblk0p2    36864   122271    85408  41.7M Microsoft basic data
/dev/mmcblk0p3   122880   208287    85408  41.7M Microsoft basic data
/dev/mmcblk0p4   208896  8597503  8388608     4G Microsoft basic data
/dev/mmcblk0p5  8597504 16986111  8388608     4G Microsoft basic data
/dev/mmcblk0p6 16986112 18349259  1363148 665.6M Linux filesystem
/dev/mmcblk0p7 18350080 61949951 43599872  20.8G Linux filesystem

Disk /dev/mtdblock0: 1.5 MiB, 1572864 bytes, 3072 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 512 bytes / 512 bytes

Disk /dev/mtdblock1: 2 MiB, 2097152 bytes, 4096 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 512 bytes / 512 bytes

Disk /dev/mtdblock2: 3 MiB, 3145728 bytes, 6144 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 512 bytes / 512 bytes

Disk /dev/mtdblock3: 128 KiB, 131072 bytes, 256 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 512 bytes / 512 bytes

Disk /dev/mtdblock4: 128 KiB, 131072 bytes, 256 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 512 bytes / 512 bytes

Disk /dev/mtdblock5: 64 KiB, 65536 bytes, 128 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 512 bytes / 512 bytes

Disk /dev/mtdblock6: 7.69 MiB, 8060928 bytes, 15744 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 512 bytes / 512 bytes

Disk /dev/mtdblock7: 1.5 MiB, 1572864 bytes, 3072 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 512 bytes / 512 bytes

installing node red packages is still possible but installing apt packages is not

thxfor the help,

Max van Eck (embedded software engineer in training )(poor intern)

jan-kiszka, 2023-09-22:
I've improved the markup in your report to make it more readable.

You built the over-the-air update-capable example image, and that has a read-only rootfs and can't take extra changes in most of its directories. Build the standard example image, and you can also live-install things. Or add things only to the directories that are in the overlay or on /home.

maxvaneck, 2023-09-23:
aah that makes sense . thx for the help.

max

jan-kiszka, 2023-09-25:
Please close if solved.

maxvaneck, 2023-09-26:
completely solved sorry for the late close

closed 2023-10-03 by jan-kiszka