siemens/meta-iot2050 #495: driver troubles

maxvaneck, 2023-10-16:
greetings,

I'm having some trouble compiling in some drivers that are by default included in the linux kernel specifically the peak_pci  driver . I managed to add the mpc251x  and the peak_usb drivers  but the peak_pci continus to elude me. So far i've added the following lines to the iot2050_defconfig_extra.cfg file. 
the lines added are 
ONFIG_CAN_PEAK_USB=y
CONFIG_CAN_MCP251X=m
CONFIG_CAN_PEAK_PCIEFD=y
CONFIG_CAN_PEAK_PCIEFD=y
Can you tell me what i'm doing wrng? or barring that is it possible to compile  the full fat kernel?

thx  for the help in advance 

Max van Eck

jan-kiszka, 2023-10-16:
I suppose this is a successor of https://support.industry.siemens.com/forum/ww/en/posts/unable-to-load-either-the-peak-pci-or-the-mcp251x-driver/305877/?page=0&pageSize=10 (at least with better notifications :wink:).

So, the peak kernel module is built and available, but you can't load? Can you share more details (error output on console and in the kernel log, dmesg)?

maxvaneck, 2023-10-16:
odprobe peak_pci
returns  below  and i don't see anything in syslog
modprobe: FATAL: Module peak_pci not found in directory /lib/modules/5.10.184-cip36

jan-kiszka, 2023-10-16:
CONFIG_CAN_PEAK_PCIEFD -> peak_pciefd
CONFIG_CAN_PEAK_PCI -> peak_pci
(not that this is always that regular, but at least in this case)

jan-kiszka, 2023-10-16:
BTW, you don't need modprobe is the right driver is available: Linux will auto-load that driver when it finds the corresponding PCI device.

maxvaneck, 2023-10-17:
Ah my bad. I don't have a pci  card on hand but  I just tested with a peak usb adapter and the driver loaded perfectly if attached on boot. I was under the assumption that if modprobe failed the driver was not included.
Thx for the help,

Max van Eck

closed 2023-10-18 by jan-kiszka