siemens/meta-iot2050 #505: Image not booting on newer IOTs with firmware >=F06?

ashleyw-iag, 2023-11-17:
Hey there, we've had some trouble with an image recently written to an SD card that didnt work on three hubs we have two of which firmware F06 and the other F07. This image however did recently for another build on another IOT (firmware yet to confirm). 

I just wondered if there are any requirements here I've missed when creating an image for certain firmwares? I noticed the example images do appear to be firmware specific on Siemens download site (with export restrictions on the newer image?). When booting, we get the STAT light show briefly then turns off with no further activity. 

Thanks, Ash.

jan-kiszka, 2023-11-17:
I'm not yet sure I get your full setup. Do you happen to have the debug console connected to your device and can share the boot logs of it?

ashleyw-iag, 2023-11-20:
Sorry for late reply, I dont, I only have ethernet for interaction with the box. We have a few of these boxes, one recently burnt with an image made from master about 1 month ago. Its an advanced box, on the sticker on the side says its firmware is F05, worked no trouble and got the OS onto the eMMC. 

We've bought a new advanced box, attempt to install on a newer firmware version and the same image will not work. The sticker on the side of this one was built much more recently, with F06 on it. Hopefully this helps?

jan-kiszka, 2023-11-20:
Recent firmware with recent image on recent hardware - that is a combination that normally works. And that's why we will need more diagnostic data to understand the problem. Specifically if you cannot make the last image from master and the last released image (pre-built) work, you will need to attache a USB-UART cable to debug port.

ashleyw-iag, 2023-11-20:
I'd love to give the prebuilt image a go, do you know where I can find that? If that doesnt work, ill try and find a USB-UART cable locally.

ashleyw-iag, 2023-11-20:
Think I've just found them, I'll update you! Thanks!

ashleyw-iag, 2023-11-21:
Prebuilt image does not work on new hardware either from my testing, I do not have a USB-UART cable so unable to debug further. We're a bit stuck, we've bought a few of the 2050 boxes, all on FS06/07, for upcoming installs that cannot be flashed to run the OS. Is it possible to downgrade this firmware as a workaround? Thanks for any support in advance.

jan-kiszka, 2023-11-21:
As the box is apparently (we have no logs...) not booting, firmware changes could only be applied via external programmers (https://github.com/siemens/meta-iot2050/tree/master/recipes-bsp/u-boot#recovering-a-bricked-device). But buying a USB-UART adapter for a Raspi is most likely simpler than setting up programmers.

@BaochengSu, any idea what could explain the observed behavior?

ashleyw-iag, 2023-11-21:
I've managed to create a debug cable with components in the workshop and can see it does go through the boot procedure, although it appears to expect input from me over the terminal to setup. This used to flash orange and let you flash to eMMC but that doesn't seem to happen - also the STAT light isn't flashing green. I had expected this light to behave as before, this could be the issue?

BaochengSu, 2023-11-21:
Sounds to me this is another #440 ?

jan-kiszka, 2023-11-21:
> Sounds to me this is another #440 ?

But this issue sounds like it reproduces all the time, unlike that sporadic reboot problem.

@ashleyw-iag, can you share the serial console log (text file)?

ashleyw-iag, 2023-11-27:
[iot2050_debug.log](https://github.com/siemens/meta-iot2050/files/13468894/iot2050_debug.log)

No worries, please see attached.

jan-kiszka, 2023-11-27:
Looks like you are booting from eMMC (because there is Industrial OS pre-installed now) while you expect to start from SD-card. You can overwrite that ordering, at least temporarily, by keeping the USER button pressed after power-up. Once the LED color changes to orange, release it, and the system will skip the eMMC.

You can make that boot ordering permanent by setting the U-Boot variable `boot_targets`, by installing what you would like to have booted on eMMC or simply by invalidating the eMMC content.

ashleyw-iag, 2023-11-27:
Worked a treat, thank you!

ashleyw-iag, 2023-11-27:
Is that a change with newer models or just bad luck?

ashleyw-iag, 2023-11-27:
not sure if this is a related issue, but still having trouble with getting the image onto eMMC, please see log attached.
[202311271456_iot2050_debug.log](https://github.com/siemens/meta-iot2050/files/13469817/202311271456_iot2050_debug.log)

After rebooting im getting a bunch of this:

...mmc1(part 0) is current device
GPT: last_usable_lba incorrect: 76F4FDE > 1da4000
find_valid_gpt: *** ERROR: Invalid GPT ***
GUID Partition Table Header signature is wrong: 0x0 != 0x5452415020494645
find_valid_gpt: *** ERROR: Invalid Backup GPT ***
GPT: last_usable_lba incorrect: 76F4FDE > 1da4000
find_valid_gpt: *** ERROR: Invalid GPT ***
GUID Partition Table Header signature is wrong: 0x0 != 0x5452415020494645
find_valid_gpt: *** ERROR: Invalid Backup GPT ***...

ashleyw-iag, 2023-11-27:
This looks like a separate issue with the image itself, closing now - thanks so much for your help!

closed 2023-11-27 by ashleyw-iag