siemens/meta-iot2050 #508: Last image and IOT PG1

rFond, 2023-12-12:
Hi, I have built a complete image from your repositories. 
When I copy the image to the device, everything is good, but when it starts, the Ethernet cards are not recognized by the Debian system.

Dec 12 13:24:01 iot2050-debian kernel : OF /bus@100000/pcie@5600000: phandle pcie-mode@4070 needs 1, found 0
Dec 12 13:24:01 iot2050-debian kernel : OF /bus@100000/pcie@5600000: phandle pcie-mode@4070 needs 1, found 0
Dec 12 13:24:01 iot2050-debian kernel : keystone-pcie 5600000.pcie: Phy link never came up

I need new firmware upgrade to use image debian 12 ?

jan-kiszka, 2023-12-12:
This is a known temporary limitation due to the TI SDK having dropped this support. We are working on bringing it back, also in mainline Linux (https://github.com/siemens/linux/commits/jan/iot2050).

rFond, 2023-12-12:
Can you tell me when the problem will be fixed?

jan-kiszka, 2023-12-13:
We hope to have something here before X-mas.

snaens, 2024-01-24:
Is this still being worked on?

BaochengSu, 2024-01-25:
Please check PR #520  , and it will be merged into master in next days if no further comments.

closed 2024-02-23 by rFond