siemens/meta-iot2050 #512: IOT2050 WITH NODE RED VERSION 3 AND NODE JS VERSION - PROBLEM WITH READING GPIO PINS

aviranp, 2023-12-14:
Hello , 

I have IOT2050 FS:05 WITH NODE RED VERSION 3 AND NODE JS VERSION 20.7 .
WORKS FINE BUT WHEN I AM TRYING TO READ GPIO PINOUT USING NODE RED INTEL GPIO NODE. 

I HAVE A A LOTS OF PROBLEMS ;

1. I CANT INSTALL IT.
2. THE NODE ONLY SUPPORT NODE JS VERSION 12. WHEN NODE RED 3 NEED MINIMUM 16.X VERSION OF NODE JS....

IF I CANT READ THE PINS I CANT WRITE ANY PROGRAM UNLESS I AM USING SIEMENS PLC. ( WORKED FINE) . 
ANY CLUE ??

THANKS

jan-kiszka, 2023-12-15:
Compile the image from the master branch (or use the output of a recent CI run), and you will get Debian 12 with NodeJS 18 with a working GPIO node.

Installing your own NodeJS version into a custom image implies that you have to take care of recompiling mraa, the foundation of node-red-node-intel-gpio, for that NodeJS yourself. Possible but not a beginner's task.

aviranp, 2023-12-31:
Hey again,

I want to compile the image but as I understood there is still a problem with the ethernet cards that they are not recognized by the Debian system? was it fixed? 

thanks,
aviran

Dec 12 13:24:01 iot2050-debian kernel : OF /bus@100000/pcie@5600000: phandle pcie-mode@4070 needs 1, found 0
Dec 12 13:24:01 iot2050-debian kernel : OF /bus@100000/pcie@5600000: phandle pcie-mode@4070 needs 1, found 0
Dec 12 13:24:01 iot2050-debian kernel : keystone-pcie 5600000.pcie: Phy link never came up

jan-kiszka, 2024-01-02:
Ethernet on PG1 needs some patches here, PG2, M.2 etc are fine already. For PG1, we are waiting for more feedback on https://lore.kernel.org/netdev/20231219174548.3481-1-diogo.ivo@siemens.com/.