siemens/meta-iot2050 #517: SWUpdate automate complete_update.sh

SCordibella, 2024-01-11:
Dear All, I have a question about the `complete_update.sh` script.
The goal is to automate as much as possible the swupdate update process in terms of user interaction.
For this reason I already add a reboot post update command in order to automatically reboot the process after the update is complete.

Now to finalize the update I need to run the `complete_update.sh` script, but if I don't want to impact on the user I need a way to automate this. My idea is to create a "one time" systemd service to run the script on first boot like the `expand-on-first-boot` service and run it after the `first-boot-complete` target.

Are there any concern to not automate the `complete_update.sh` script? 
Why it is not already automated? I guess that is just to let the user decide the best update approach.

Best regards,
Stefano.

jan-kiszka, 2024-01-11:
`complete_update.sh` is a placeholder for the typically deployment-specific task of identifying if the system is updatable again after having just applied an update. If you pull updates from a server, make sure that the update service doing that can reach the server, and only then confirm the update. If you are pulling the update from an external storage, and the storage media that brought the current update is still plugged, check that you can still access that media. Etc., etc. If you don't do any sanity check, you may automatically "brick" your device in the field via an unlucky update, even if the device comes up again.

SCordibella, 2024-01-11:
Thank you @jan-kiszka for the answer.

Are there any documentation about the intended use of complete_update.sh or more generally speaking about the EFI Boot Guard for iot2050?

SCordibella, 2024-01-11:
Just because we are speaking about the `swupdate-complete-update-helper_0.1.bb`...
I don't think that the comment is correct in the `do_install()`:

```
do_install() {
    # add board status led service
    install -v -d ${D}/usr/bin
    install -v -m 644 ${WORKDIR}/complete_update.sh ${D}/usr/bin/complete_update.sh
}
```

jan-kiszka, 2024-01-11:
Yes... thanks for pointing out.

And we unfortunately have no real guiding documents on the SWUpdate example integration. That is expert land so far, although we plan to improve that via upstream (Civil Infrastructure Platform, isar-cip-core).

jan-kiszka, 2024-01-11:
I've pushed at least some explanations from this thread into our README, see #518.

SCordibella, 2024-01-11:
Thank you @jan-kiszka for your help. 

I think that we can close this issue.

closed 2024-01-11 by SCordibella