siemens/meta-iot2050 #524: Including .NET Core in the image.

Int3rC3pt0r, 2024-02-08:
Hello,

I am trying to include `dotnet-core 6.0.26` in the image for my iot2050.
Therefore I tried to create a recipe based on Robert Dunkleys [meta-dotnet-core](https://github.com/RDunkley/meta-dotnet-core) like this, where dotnet core is directly downloaded form microsoft:

dotnet-core_6.0.26.bb:

```
###################################################################################################
# Contains the recipe to download the release binaries from Microsoft for the version
# 6.0.26 .Net runtime.
# Copyright Richard Dunkley 2024
# Auto-generated using meta_dotnet_core_gen
###################################################################################################

inherit dpkg-raw

SRC_URI = "https://download.visualstudio.microsoft.com/download/pr/226d8ab0-8daf-47b1-80cb-a9b68badb28b/a4425bebee91775ce41a50aca80e09c1/dotnet-runtime-6.0.26-linux-arm64.tar.gz;subdir=dotnet-${PV}"

SRC_URI[md5sum] = "4ebc14777370ce6464e121b899406218"
SRC_URI[sha256sum] = "17635cee12f12b28e642e6a6fb8663bfd86b653955e4775545c08c77bb7871db"

require dotnet-core_6.x.x.inc

```

dotnet-core_6.x.x.inc:

```
###################################################################################################
# Contains additional parameters for the recipe to download the release binaries from Microsoft
# pertaining to all 6.x.x versions.
# Copyright Richard Dunkley 2021
###################################################################################################
DESCRIPTION = "This does not build this version, it only pulls the binaries from Microsoft"
BUGTRACKER = "https://github.com/dotnet/core/issues"
SECTION = "runtime"

S = "${WORKDIR}/dotnet-${PV}"

bindir="/usr/bin"
datadir="/usr/share"

# FIXME: patchelf-native must be removed if hack in do_install:append not required
DEBIAN_BUILD_DEPENDS = "patchelf, liblttng-ust-dev, zlib1g-dev, libicu-dev, libssl-dev, lttng-tools"
DEBIAN_DEPENDS = "zlib1g"

do_configure[noexec] = "1"
do_compile[noexec] = "1"

FILES:${PN} += " \
	${datadir}/dotnet \
	/usr/local/share/dotnet \
	"
SRC_URI += " \
    file://10_dotnet_env.sh \
	file://postinst"

do_install:prepend () {
  bbnote "start dotnet"
		install -d ${D}${bindir}
		install -d ${D}${datadir}/dotnet/
		install -m 0755 ${S}/dotnet ${D}${datadir}/dotnet/
		install -m 0644 ${S}/LICENSE.txt ${D}${datadir}/dotnet/
		install -m 0644 ${S}/ThirdPartyNotices.txt ${D}${datadir}/dotnet/

	bbnote "start dotnet fxr"
		install -d ${D}${datadir}/dotnet/host/fxr/${PV}/
		install -m 0755 ${S}/host/fxr/${PV}/libhostfxr.so ${D}${datadir}/dotnet/host/fxr/${PV}/

	bbnote "start Microsoft.NETCore.App"
		install -d ${D}${datadir}/dotnet/shared/Microsoft.NETCore.App/${PV}/
		
		install -m 0644 ${S}/shared/Microsoft.NETCore.App/${PV}/.version ${D}${datadir}/dotnet/shared/Microsoft.NETCore.App/${PV}/.version
		install -m 0644 ${S}/shared/Microsoft.NETCore.App/${PV}/createdump ${D}${datadir}/dotnet/shared/Microsoft.NETCore.App/${PV}/createdump
		for file in ${S}/shared/Microsoft.NETCore.App/${PV}/*.so; do
			install -m 0755 "$file" ${D}${datadir}/dotnet/shared/Microsoft.NETCore.App/${PV}/
		done
		for file in ${S}/shared/Microsoft.NETCore.App/${PV}/*.dll; do
			install -m 0644 "$file" ${D}${datadir}/dotnet/shared/Microsoft.NETCore.App/${PV}/
		done
		for file in ${S}/shared/Microsoft.NETCore.App/${PV}/*.json; do
			install -m 0644 "$file" ${D}${datadir}/dotnet/shared/Microsoft.NETCore.App/${PV}/
		done

		install -d ${D}/usr/local/share

}

do_install:append() {   
	# cd ${D}/usr/local/share
	# ln -s ../../share/dotnet ${D}/usr/local/share

	cd ${D}${bindir}
	ln -s ../share/dotnet/dotnet ${D}${bindir}/dotnet

    # FIXME: must be removed if the liblttng-ust library issue was fixed
    # Hack to fix liblttng-ust dependency issues
     patchelf --remove-needed liblttng-ust.so.0 ${D}${datadir}/dotnet/shared/Microsoft.NETCore.App/${PV}/libcoreclrtraceptprovider.so
}
```

The problem is that when running the kas-container, I get the error `/build/tmp/work/iot2050-debian-arm64/dotnet-core/6.0.26-r0/temp/run.do_install.142: 101: patchelf: not found`, even though patchelf is included in the dependencies.
When i run the build without patchelf i get the error `dpkg-shlibdeps: error: cannot find library liblttng-ust.so.0 needed by debian/dotnet-core/usr/share/dotnet/shared/Microsoft.NETCore.App/6.0.26/libcoreclrtraceptprovider.so (ELF format: 'elf64-littleaarch64' abi: '020100b700000000'; RPATH: '')`.

I am still new to kas/isar/yocto, so any help and insight would be appreciated.

Thanks in advance,
Thomas

jan-kiszka, 2024-02-08:
You were looking at a Yocto image, but meta-iot2050 is Isar, i.e. bitbake for plumbing Debian packages together. If Mono is sufficient as well: `apt update; apt install mono-runtime`. Otherwise, a bit more effort might be needed to port the logic from the Yocto recipes into a Debian package via Isar. But as MS .NET is binary only, you just need a `dpkg-raw` recipe that fetches the bins and installs them.

jan-kiszka, 2024-02-08:
Oh, you already played with that (just read "yocto" before): Debian has https://packages.debian.org/bookworm/liblttng-ust1, that may resolve the missing dep. There is also https://learn.microsoft.com/en-us/dotnet/iot/deployment which suggests that it just works on Raspian which is basically Debian.

Int3rC3pt0r, 2024-02-08:
Thanks for the quick reply! Using liblttng-ust1 sadly doesn't work either (still gives the same error with liblttng-ust.so.0). Also Mono isn't really an option for me. The thing is that the build already worked with Debian 10 before the move to liblttng-ust1 in Debian 12. My main problem is that the command `patchelf` is not recognized even though it is included in the build dependencies.

Thanks again,
Thomas

SCordibella, 2024-02-08:
Hi @Int3rC3pt0r, I made a similar recipe, but for aspnetcore-runtime with the following deps:
```
DEBIAN_DEPENDS = "libc6, libgcc-s1, libgssapi-krb5-2, libicu72, liblttng-ust1, libssl3, libstdc++6, zlib1g"
```
In addition I create a `rules` file with the following content:
```
#!/usr/bin/make -f
%:
	dh $@

override_dh_strip:
	true

override_dh_shlibdeps:
	true

```
Hope this helps.

Best regards,
Stefano.

jan-kiszka, 2024-02-09:
Thanks for reminding! That was discussed in https://github.com/siemens/meta-iot2050/issues/439.

Int3rC3pt0r, 2024-02-09:
@SCordibella this seems to have done the trick and the build ran trough! Thank you very much 😁.

closed 2024-02-23 by Int3rC3pt0r