siemens/meta-iot2050 #525: Debian 12 / ttyX30

rFond, 2024-02-21:
Hello, with the latest version of Debian (FS03), the ttyX30 port no longer exists. To fix the issue, I had to modify the /etc/udev/rules.d/20-create-symbolic-link-for-serial-port.rules file.

The kernel is no longer correct:
udevadm info --attribute-walk --path=$(udevadm info --query=path --name=/dev/ttyUSB0)

looking at parent device '/devices/platform/bus@100000/4000000.dwc3/4010000.usb/xhci-hcd.0.auto/usb1/1-1/1-1.4/1-1.4:1.0':
KERNELS=="1-1.4:1.0"
SUBSYSTEMS=="usb"
DRIVERS=="cp210x"
ATTRS{authorized}=="1"
ATTRS{bAlternateSetting}==" 0"
ATTRS{bInterfaceClass}=="ff"
ATTRS{bInterfaceNumber}=="00"
ATTRS{bInterfaceProtocol}=="00"
ATTRS{bInterfaceSubClass}=="00"
ATTRS{bNumEndpoints}=="02"
ATTRS{supports_autosuspend}=="1"

So in my new file :
SUBSYSTEM=="tty", SUBSYSTEMS=="usb", KERNELS=="1-1.4:1.0", DRIVERS=="cp210x", SYMLINK+="ttyX30"
SUBSYSTEM=="tty", KERNEL=="ttyS2", DRIVERS=="omap8250", SYMLINK+="ttyX30"

jan-kiszka, 2024-02-21:
Good catch! Do you want to write this fix as patch, or should we?

rFond, 2024-02-21:
I'm happy for you to do it. I'm not very comfortable with Git.
Thank's

jan-kiszka, 2024-02-21:
- fork this repo on github
- `git clone ...`
- `git checkout -b ttyX30-fix` (or whatever name you want to choose for the branch
- edit file
- `git commit -asv`
- `git push origin ttyX30-fix`
- click on the link shown locally to create a merge request

That's a rough plan, and I probably forgot some detail now. Drop questions then. And the rest we can sort out in the merge request.

rFond, 2024-02-21:
Sorry but I won't be able to correct this problem. I don't have the technical knowledge to do this

jan-kiszka, 2024-02-22:
No problem, will take over.

jan-kiszka, 2024-02-22:
It turned out to be more complex: The issue was rooted in the wiring differences between PG1 and PG2 devices, not in the software version. Please give the proposed solution a try.

commit 1c44cc4c (2024-02-23): customizations-example: Adjust ttyX30 rule to Advanced PG1

We were missing the differences between PG1 and PG2-type devices, namely
that the USB topology was swapped between both.

Closes #525
closed 2024-02-22 by rFond