siemens/meta-iot2050 #527: IOT2050 - Node-red to Adafruit  (MQQT not connecting)

davidstuart97, 2024-03-01:
Hi,

I'm in need of assistance in resolving why I can't connect to Adafruit through Node-RED. I've successfully connected to Adafruit using the same flows on Node-RED from another computer.

On my IOT2050 device:

Node-RED is operational, successfully retrieving data from a PLC.
Mosquitto: I've added a listener and checked its status, which seems okay.
However, I keep encountering this error message: '26 Feb 18:15:29 - [info] [mqtt-broker:Adafruit] Connection failed to broker: mqtt://io.adafruit.com:1883'

The password and username for Adafruit is correct as used on another computer.

What could I be missing?

jan-kiszka, 2024-03-01:
Can you do `telnet io.adafruit.com 1883` on the IOT2050, i.e. reach that broker at all via its Internet connection?

davidstuart97, 2024-03-02:
I've tried this:-
2 Mar 10:37:27 - [info] [mqtt-broker:Adafruit] Connection failed to broker: mqtt://io.adafruit.com:1883
2 Mar 10:37:42 - [info] [mqtt-broker:Adafruit] Connection failed to broker: mqtt://io.adafruit.com:1883
^C2 Mar 10:37:43 - [info] Stopping flows
2 Mar 10:37:43 - [info] Stopped flows
root@iot2050-dave://# sudo telnet io.adafruit.com1883
sudo: unable to resolve host iot2050-dave: Temporary failure in name resolution
sudo: telnet: command not found
root@iot2050-dave://# ^C
root@iot2050-dave://# sudo apt-get install telnet
sudo: unable to resolve host iot2050-dave: Temporary failure in name resolution
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
E: Unable to locate package telnet
root@iot2050-dave://# root@iot2050-dave://# sudo apt-get install telnet
sudo: unable to resolve host iot2050-dave: Temporary failure in name resolution
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
E: Unable to locate package telnet
root@iot2050-dave://#

rFond, 2024-03-05:
Hello, check your internet connection.
If you use the 2 ethernet port you can check the box "Never use this network for default route" into iot2050setup / Networking / Edit a connection / "your network interface"

davidstuart97, 2024-03-06:
Hi rFord,

On initial Setup I changed the network address to give access to the unit.  I've managed to connect the unit by using Mobaxterm (like putty) and have established a connection.  I also changed the default port address and still I can't seem to connect with Adafruit using Mqqt protcol.

root@iot2050-dave://etc/mosquitto# Active Internet connections (servers and established)
Proto Recv-Q Send-Q Local Address           Foreign Address         State
tcp        0      0 0.0.0.0:22              0.0.0.0:*               LISTEN
tcp        0      0 0.0.0.0:1534            0.0.0.0:*               LISTEN
tcp        0      0 192.168.1.173:22        192.168.1.55:60057      ESTABLISHED
tcp        0      0 192.168.1.170:22        192.168.1.55:60050      ESTABLISHED
tcp        0      0 192.168.1.173:22        192.168.1.55:60054      ESTABLISHED
tcp        0   4240 192.168.1.170:22        192.168.1.55:60049      ESTABLISHED
tcp6       0      0 :::22                   :::*                    LISTEN
udp        0      0 0.0.0.0:1534            0.0.0.0:*
raw6       0      0 :::58                   :::*                    7
Active UNIX domain sockets (servers and established)
Proto RefCnt Flags       Type       State         I-Node   Path
unix  2      [ ]         DGRAM                    1829     /run/user/0/systemd/notify
unix  2      [ ACC ]     STREAM     LISTENING     1832     /run/user/0/systemd/private
unix  3      [ ]         DGRAM                    1409     /run/systemd/notify
unix  2      [ ACC ]     STREAM     LISTENING     1412     /run/systemd/private
unix  2      [ ACC ]     STREAM     LISTENING     1414     /run/systemd/userdb/io.systemd.DynamicUser
unix  2      [ ACC ]     STREAM     LISTENING     1415     /run/systemd/io.system.ManagedOOM
unix  2      [ ]         DGRAM                    1419     /run/systemd/journal/syslog
unix  12     [ ]         DGRAM                    1423     /run/systemd/journal/dev-log
unix  7      [ ]         DGRAM                    1425     /run/systemd/journal/socket
unix  2      [ ACC ]     STREAM     LISTENING     1427     /run/systemd/journal/stdout
unix  2      [ ACC ]     SEQPACKET  LISTENING     1429     /run/udev/control
unix  2      [ ACC ]     STREAM     LISTENING     2302     /run/systemd/journal/io.systemd.journal
unix  2      [ ACC ]     STREAM     LISTENING     6607     /run/dbus/system_bus_socket
unix  2      [ ACC ]     STREAM     LISTENING     1532     /run/uuidd/request
unix  3      [ ]         STREAM     CONNECTED     1834
unix  2      [ ]         DGRAM                    2391
unix  2      [ ]         STREAM     CONNECTED     1796
unix  2      [ ]         DGRAM                    1675
unix  3      [ ]         DGRAM                    1411
unix  3      [ ]         DGRAM                    1830
unix  3      [ ]         DGRAM                    1410
unix  2      [ ]         DGRAM                    7125
unix  3      [ ]         DGRAM                    1831
unix  3      [ ]         DGRAM                    2398
unix  2      [ ]         STREAM     CONNECTED     8341
unix  3      [ ]         STREAM     CONNECTED     1453
unix  3      [ ]         DGRAM                    2397
unix  3      [ ]         STREAM     CONNECTED     1478     /run/systemd/journal/stdout
unix  3      [ ]         STREAM     CONNECTED     2823     /run/dbus/system_bus_socket
unix  2      [ ]         DGRAM                    1462
unix  3      [ ]         DGRAM                    1465
unix  3      [ ]         STREAM     CONNECTED     1541
unix  3      [ ]         DGRAM                    1466
unix  2      [ ]         DGRAM                    2307
unix  2      [ ]         STREAM     CONNECTED     7118
unix  3      [ ]         STREAM     CONNECTED     6639     /run/dbus/system_bus_socket
unix  3      [ ]         DGRAM                    2396
unix  3      [ ]         STREAM     CONNECTED     1545
unix  3      [ ]         DGRAM                    2395
unix  3      [ ]         STREAM     CONNECTED     6613     /run/systemd/journal/stdout
unix  2      [ ]         DGRAM                    2493
unix  3      [ ]         STREAM     CONNECTED     472
unix  3      [ ]         STREAM     CONNECTED     6636
unix  3      [ ]         STREAM     CONNECTED     2377     /run/systemd/journal/stdout
unix  3      [ ]         STREAM     CONNECTED     7187     /run/dbus/system_bus_socket

I've ordered up a UART to usb but will keep  looking for a solution.
Thanks for your support

davidstuart97, 2024-03-19:
I seem to have an issue with Mqqt to load up to give remote access.

I've added : -
listener 1883
allowed anonymous

But it still doesn't;t connect

root@iot2050-dave://etc/mosquitto# mosquitto
1710843904: mosquitto version 2.0.11 starting
1710843904: Using default config.
1710843904: Starting in local only mode. Connections will only be possible from clients running on this machine.
1710843904: Create a configuration file which defines a listener to allow remote access.
1710843904: For more details see https://mosquitto.org/documentation/authentication-methods/
1710843904: Opening ipv4 listen socket on port 1883.
1710843904: Opening ipv6 listen socket on port 1883.
1710843904: mosquitto version 2.0.11 running