siemens/meta-iot2050 #529: Next release for IOT2050

bramroelandts, 2024-03-06:
Hi,

The latest release tag on GitHub (V01.03.01) dates back from September 2022. Since then, a lot of work has happened on the master branch (a move to Debian 12 and NodeJS 18, notably). We want to upgrade to a newer release, but are waiting for a new release to come out. We want to avoid simply compiling the latest master at a random point in time, since we risk ending up with software in the middle of some transition or new feature.

Are there any plans for a new release tag on GitHub which is considered stable and complete? I know the master is supposed to be stable, but not necessarily complete (see the M.2 transition, Debian 12 transition etc).

Any feedback would be appreciated.
Thanks

BaochengSu, 2024-03-07:
Hi,

Currently we are very close to the release of the next V-tag (should be V01.04.01), development & testing are almost done, meanwhile we need go through some internal process to publish it.

Can't promise any datetime here, but I think it will not take long to tag the new release version.

Anyway, the current master head (https://github.com/siemens/meta-iot2050/commit/83cd49983e7c4d17722bc4495ba35964e22f5c04) is the current release candidate, if no further issue reported.

bramroelandts, 2024-03-07:
Thank you for this feedback, it is much appreciated. We will await the next release then once the internal proceedings are rounded up. Thanks for the great work on this product.

closed 2024-03-07 by bramroelandts