siemens/meta-iot2050 #536: Kernel Panic when doing anything Network related

Int3rC3pt0r, 2024-03-29:
Hello there,

I recently upgraded to a newer meta-iot2050 commit in my image and since then I more often than not get a kernel panic when running network related commands like `sudo systemctl restart networking` or `sudo ifdown eno1`. This seems to be connected with dhcp since running `sudo ifdown eno2` on my static network interface works perfectly fine. This also doesn't happen consistently because sometimes it works as if there are no problems at all.

The interfaces in my image are configures like this:
- /etc/network/interfaces
```
# interfaces
source /etc/network/interfaces.d/*

allow-hotplug eno1
iface eno1 inet dhcp

# The loopback network interface
auto lo
iface lo inet loopback
```
- /etc/network/interfaces.d/static_interface
```
allow-hotplug eno2
iface eno2 inet static
 address 192.168.200.10
 netmask 255.255.255.0
```

The output of the kernel panic: [kernel_panic.txt](https://github.com/siemens/meta-iot2050/files/14805649/kernel_panic.txt)

Any ideas as to why this happens would be appreciated!

Int3rC3pt0r, 2025-01-30:
Disabling the NetworkManager service fixed the issue.

rFond, 2026-10-05:
This is the same crash we hit on IOT2050 Advanced (kernel 6.12.90-cip24): `k3_knav_desc_pool size 65536 != avail 0` in `emac_ndo_stop()`, then `kernel BUG at lib/genalloc.c:255`, exactly as in the attached trace.

Root cause: `k3_udma_glue_reset_rx_chn()` returns the free descriptor queue to the pool, but it only resets the completion ring. RX descriptors already completed by the DMA but not yet consumed by NAPI are lost, so destroying the descriptor pool in `ndo_stop()` finds it not full and hits `BUG()`. It is sporadic in normal operation and systematic when the port's RX path is stalled (see #368), which fits "sometimes it works".

The fix (stop NAPI before the reset, then reclaim the completed descriptors) and our bench results are posted in https://github.com/siemens/meta-iot2050/issues/368#issuecomment-5999808961 (patch `0002-net-ti-icssg-prueth-reclaim-completed-RX-descriptors-on-stop.patch`). With it: 50 out of 50 stop/start cycles of the interface under traffic, no WARN/BUG.