siemens/meta-iot2050 #540: IOT & Modem sometimes not working well

edu-sistec, 2024-04-16:
I have IOT2050 Basics FS:04 and 4G modem Huawei ME909s-821 .
Everything is running with Example Image V1.3.1.

The operation is correct. But after a few days working correctly on some devices, not all IOTs, when restarting the OS, the ModemManager application stops starting properly and does not detect the modem, although the OS has detected the modem correctly and the NetworkManager has runned the ModemManager application.

Looking at the syslogs, the error starts when a line that should appear does not appear, compared to other devices that do work:

ModemManager[342]: opening device...

From this line, which no longer appears, the following Modemmanager lines no longer appear either. Then the IOT fails to connect to the internet.

I restart the OS but the Modemmanager never starts well.

The only solution I have is to remove voltage and apply it again, then the ModemManager does start fine.

I would really appreciate it if someone could shed some light on this problem, I don't know where to start looking to fix it.

If anyone has been through this or has any ideas I would appreciate any feedback as I am pretty lost.

thanks in advance,

Edu

edu-sistec, 2024-04-23:
Hello,
Finally, the IOT I have tested has caused the error.
As you can see in the attached image, I've had ModemManager give me the modems it found and it doesn't find any.
I have restarted the ModemManager application and the NetworkManager without fixing the problem. But actually doing 'lsusb' the OS finds the modem.
I finally solved it by restarting the IOT, but this is not an acceptable solution for IOTs I have in production.

Does anyone have an idea for a better solution?
Thanks in advance.
[image]

andyliao5555, 2024-05-06:
please refer and try below links:
https://draisberghof.de/usb_modeswitch/bb/viewtopic.php?f=3&t=2997&p=19977#p19977
https://www.draisberghof.de/usb_modeswitch/bb/viewtopic.php?f=2&t=3034&p=19976

edu-sistec, 2024-05-07:
Hi andyliao,

Thank you very much for this contribution. I was really troubled with this problem.

I found that the problem comes with Huawei ME909s-821 Version 2 modems.

With the input you gave me, I have already done some tests with good results, I have made modifications in a file.

Now I want to wait a few days of testing since the errors were coming out after 7 days without removing voltage from the IOT

Thanks.