siemens/meta-iot2050 #542: secure-boot image not booting in iot2050 PG2 Advanced.

rakeshk7097, 2024-04-18:
Hi team,

I am using the IOT2050 PG2 advanced boad and trying to boot the secure boot image(iot2050-image-swu-example-iot2050-debian-iot2050.wci) in there.

I have used the below command to generate the signed image.

1. Using the below command to generate the signed firmware image, 
./kas-container build kas-iot2050-boot.yml:kas/opt/secure-boot.yml:kas/opt/otpcmd/key-provision.yml

2.  Using below command to generate the signed UKI image.
 ./kas-container build kas-iot2050-swupdate.yml:kas/opt/secure-boot.yml

Currently i am using the default key which is stored in below location.
meta-iot2050/recipes-devtools/secure-boot-secrets/files/

I have flashed the signed firmware image and now trying to boot the flashed wic image from USB and getting below error at boot time.

`Begin: Loading essential drivers ... done.
Begin: Running /scripts/init-premount ... done.
Begin:	 ... Begin: Running /scripts/local-top ... Device /dev/mmcblk1p4 is not a valid VERITY device.
Device /dev/sda4 is not a valid VERITY device.
Device /dev/sda7 is not a valid VERITY device.
Device /dev/sda6 is not a valid VERITY device.
Begin: Waiting for IMAGE_UUID= ... [   11.503342] random: crng init done
[   21.423475] sd 0:0:0:0: [sda] tag#0 UNKNOWN(0x2003) Result: hostbyte=0x07 driverbyte=DRIVER_OK cmd_age=0s
[   21.423540] usb 3-1.1: USB disconnect, device number 4
[   21.433092] sd 0:0:0:0: [sda] tag#0 CDB: opcode=0x28 28 00 00 00 28 00 00 00 08 00
[   21.433105] I/O error, dev sda, sector 10240 op 0x0:(READ) flags 0x80700 phys_seg 1 prio class 2
[   21.454973] device offline error, dev sda, sector 10240 op 0x0:(READ) flags 0x80700 phys_seg 1 prio class 2
[   21.464929] Buffer I/O error on dev sda1, logical block 1024, async page read

done.
Device  does not exist or access denied.
Can't open verity rootfs - continuing will lead to a broken trust chain!
Rebooting automatically due to panic= boot argument`

Attached the full boot log file.
[meta-iot2050_boot_log.txt](https://github.com/siemens/meta-iot2050/files/15022549/meta-iot2050_boot_log.txt)

@jan-kiszka , could you please give your suggestions on this error.

huaqianli, 2024-04-19:
@rakeshk7097 it appears there's been a misunderstanding. We utilize eMMC RPMB for UEFI key storage, so the secure boot path you've followed seems incorrect. Here are the correct steps:

1. Prepare the UEFI key authenticated data, ensuring it's associated with the owner GUID.
2. Enroll the UEFI keys.
3. Flash the signed firmware and image to the eMMC.
4. Reboot the system. This will enable secure boot.

In your case, you've flashed the image onto a USB disk instead of the eMMC. This doesn't align with the trust chain requirements. Please follow the steps above to ensure a secure boot process.

More details please refer to the secure boot manual: IOT2050_Secure_Boot_en.pdf - IOT2050 Secure Boot Example Implementation.

rakeshk7097, 2024-04-19:
Hi @huaqianli , thanks for your response.

Now we have flashed the wic image into sd card and boot the device with mmc0. 
Attached the boot log. 
Looks like it's not able to get the rootfs partition's UUID with we are giving in wic file as 
```
part --source rawcopy --sourceparams "file=${IMAGE_FULLNAME}.verity" --align 1024 --fixed-size 4G --uuid "${ABROOTFS_PART_UUID_A}"
part --source empty --align 1024 --fixed-size 4G --uuid "${ABROOTFS_PART_UUID_B}"
```
If I am not wrong .verity will be treated as rootfs to boot.
Here :ABROOTFS_PART_UUID_A is assigned in the 
```
conf/distro/iot2050-debian.conf
```
ABROOTFS_PART_UUID_A ?= "fedcba98-7654-3210-cafe-5e0710000001"

Should this ABROOTFS_PART_UUID_A value which seems as UUID of rootfs be equal to what UUID is generated in image.verity.env
```
/build/tmp/deploy/images/iot2050$ cat iot2050-image-swu-example-iot2050-debian-iot2050.verity.env
UUID=81fffc2e-86d7-41bb-8ecc-c9abb8704725
HASH_TYPE=1
DATA_BLOCKS=745260
DATA_BLOCK_SIZE=1024
HASH_BLOCKS=24042
HASH_BLOCK_SIZE=1024
HASH_ALGORITHM=sha256
SALT=e8874e91e63aed90f2e080a9a0bfc4d12782ea3830cefdcbd2f70fb2a7faea08
ROOT_HASH=2d77c3e6938b38a4ecb9983af23ed779356314c402545fe3444b0a6b11cbbf32
HASH_OFFSET=763146240
```
Could you please give your inputs on these?

huaqianli, 2024-04-23:
@rakeshk7097 I couldn't understand your problem completely without the complete context. Since you were following the old version of secure boot documentation. Perhaps it would be beneficial to revisit the issue after obtaining the latest documentation release.

sbobade, 2024-04-23:
@huaqianli sure no problem. we will wait for the new document to be available for use.

rakeshk7097, 2024-07-09:
Secure boot is working fine now after flashing the signed u-boot. 

thanks @huaqianli for your support and help on this. 

Closing this ticket.

closed 2024-07-09 by rakeshk7097