siemens/meta-iot2050 #553: Swupdate watchdog timeout not working

Int3rC3pt0r, 2024-06-25:
When running swupdate manually with for example `swupdate -i image.swu`, the update is installed on the unused partition (e.g. `mmcblk1p5`). On the following reboot (booting from `mmcblk1p5`) watchdog should start, giving me one minute to confirm the successful installation by running the `complete_update.sh` script. If this is not done within one minute (and please correct me if I'm wrong), watchdog should restart the system, which should result with the system booting into the previous partition (`mmcblk1p4`). This all works perfectly fine except for watchdog triggering the reboot.

The boot logs mention some issue with setting the timeout, but the default timeout should already be one minute. When using `bg_printenv` a timeout of zero seconds is shown.

From the boot log ([whole_boot.log](https://github.com/user-attachments/files/15965863/whole_boot.log)):
```
[  855.558516] watchdog: watchdog0: nowayout prevents watchdog being stopped!
[  855.565445] watchdog: watchdog0: watchdog did not stop!
[  855.608032] systemd-shutdown[1]: Using hardware watchdog 'K3 RTI Watchdog', version 0, device /dev/watchdog0
[  855.617961] systemd-shutdown[1]: Modifying watchdog timeout is not supported, reusing the programmed timeout.
[  855.627959] systemd-shutdown[1]: Watchdog running with a timeout of 1min.
[  855.939774] watchdog: watchdog0: nowayout prevents watchdog being stopped!
[  855.946646] watchdog: watchdog0: watchdog did not stop!
WDT:   Not starting watchdog@40610000
Watchdog started, timeout 60000 ms
Starting C:BOOT1:linux.efi with watchdog set to 0 seconds ...
[    0.000000] Kernel command line: console=ttyS3,115200n8 earlycon=ns16550a,mmio32,0x02810000 rootwait watchdog.handle_boot_enabled=0
[    2.168983] watchdog: watchdog0 running and kernel based pre-userspace handler disabled
[    5.557697] systemd[1]: Using hardware watchdog 'K3 RTI Watchdog', version 0, device /dev/watchdog0
[    5.566810] systemd[1]: Modifying watchdog timeout is not supported, reusing the programmed timeout.
[    5.576207] systemd[1]: Watchdog running with a timeout of 1min.
```

bg_printenv output after executing swupdate:
```
----------------------------
 Config Partition #0 Values:
in_progress:      no
revision:         2
kernel:           C:BOOT0:linux.efi
kernelargs:
watchdog timeout: 0 seconds
ustate:           0 (OK)

user variables:
----------------------------
 Config Partition #1 Values:
in_progress:      no
revision:         3
kernel:           C:BOOT1:linux.efi
kernelargs:
watchdog timeout: 0 seconds
ustate:           2 (TESTING)

user variables:
```

start_watchdog is also configured in the U-Boot environment:
```
$ sudo fw_printenv | grep watchdog
bootcmd=run start_watchdog; run distro_bootcmd
start_watchdog=if test ${watchdog_timeout_ms} -gt 0; then wdt dev watchdog@40610000; wdt start ${watchdog_timeout_ms}; echo Watchdog started, timeout ${watchdog_timeout_ms} ms; fi
watchdog_timeout_ms=60000
```

Is it intendet to only switch back to the original partition when triggering the reset manually?

jan-kiszka, 2024-06-25:
The watchdog phase starting with the firmware (or, on x86 devices, with EFI Boot Guard) is there to guard kernel boot up to the point that systemd can take over monitoring of the system. That means if your post-update health check fails, you actively need to restart the system. If the service that is supposed to check that is failing, systemd is your first line of defense. Only if systemd crashes, the hw watchdog comes into play again.

Yes, this also needs documentation, likely in isar-cip-core... @gylstorffq @fmoessbauer

Int3rC3pt0r, 2024-06-25:
Thanks for the clarification!

closed 2024-06-25 by Int3rC3pt0r