siemens/meta-iot2050 #554: board-bootup-conf and iot2050setup need permission

applea9, 2024-06-28:
Hi,

in the newest v1.4.1, board-bootup-conf and iot2050setup miss the executable permission.

you can check the .bb file : [https://github.com/siemens/meta-iot2050/blob/V01.04.01/recipes-app/board-conf-tools/board-conf-tools_0.1.bb](https://github.com/siemens/meta-iot2050/blob/V01.04.01/recipes-app/board-conf-tools/board-conf-tools_0.1.bb)

`do_install() {
    install -v -d ${D}/usr/bin/
    install -v -d ${D}/etc
    install -v -d ${D}/lib/systemd/system/
    # add board bootup configuration service
    install -v -m 644 ${WORKDIR}/board-bootup-configuration.service ${D}/lib/systemd/system/

    cp -rf  ${WORKDIR}/board-conf-tools ${D}/etc
    ln -sf /etc/board-conf-tools/iot2050setup.py   ${D}/usr/bin/iot2050setup
    ln -sf /etc/board-conf-tools/board-bootup-conf.py   ${D}/usr/bin/board-bootup-configuration
}`

huaqianli, 2024-06-28:
In my test, both of them worked fine and have the executable permissions, as follows:
[image]

Could you please share more detailed information about this issue?

jan-kiszka, 2024-06-29:
This scripts are clearly installed into the wrong place (/etc is wrong, should go directly into /usr/bin, without pointless links). But that has been like that a bit longer.