siemens/meta-iot2050 #555: rootfs encryption issue on advanced PG2  device

rakeshk7097, 2024-07-09:
Hi all, 

I am trying to encrypt the rootfs partition on PG2 advanced device with adding below line in `cip-core/recipes-initramfs/initramfs-crypt-hook/initramfs-crypt-hook.bb` as mentioned in doc [isar-cip-core](https://gitlab.com/cip-project/cip-core/isar-cip-core/-/blob/master/doc/README.tpm2.encryption.md?ref_type=heads)

`CRYPT_PARTITIONS ??= "${ABROOTFS_PART_UUID_A}::reencrypt ${ABROOTFS_PART_UUID_B}::reencrypt"`

But it's not working and getting below error while booting the board.

`Begin: Mounting root file system ... Begin: Running /scripts/local-top ... ERROR:tcti:src/tss2-tcti/tcti-device.c:452:Tss2_Tcti_Device_Init() Fai 
ERROR:tcti:src/tss2-tcti/tctildr-dl.c:154:tcti_from_file() Could not initialize TCTI file: device 
ERROR:tcti:src/tss2-tcti/tctildr.c:428:Tss2_TctiLdr_Initialize_Ex() Failed to instantiate TCTI 
ERROR: Could not load tcti, got: "device:/dev/tpm*"
No tpm device exists or supports pcr_hash 'sha256' or 'ecc' - cannot create a encrypted device!`

I am using the cip-core commit `6c24d487e3df0107756487e8a0dfd0a880644b02` and isar commit `d2d3b3e94874d62d48c0cafb99d`.

Please give inputs/suggestions on this.

thanks!

sbobade, 2024-07-09:
with same setup we have now Secure  Boot working well https://github.com/siemens/meta-iot2050/issues/542 along with encryption for default partitions i. /var and ii /home  works well. only issue we are seeing with now `rootfs` partition encryption.

rakeshk7097, 2024-11-18:
Issue is not reproducible now, and now working as expected, so closing the ticket, thanks.

closed 2024-11-18 by rakeshk7097