siemens/meta-iot2050 #558: Upstream kernel 6.x vs default iot2050-6.1.54

SCordibella, 2024-08-07:
Dear All, 
    I just want to know the the difference between the default kernel for iot2050 (6.1.54-cip6) and the upstream one.
I see that the upstream is more updated (6.10 at the time of writing) and I guess to choose that kernel since it contains security patch like the one for CVE 2024-1086 .

Are there other significant changes between the two kernels?

Best regards,
Stefano.

BaochengSu, 2024-08-08:
Dear Stefano,

According to [1], the "upstream" kernel you mentioned is the "mainline tree":

> Mainline tree is maintained by Linus Torvalds. It's the tree where all new features are introduced and where all the exciting new development happens. New mainline kernels are released every 9-10 weeks

This tree is where the development happens and is considered "unstable"; hence, it should not be used for production purposes as new feature introductions may bring breakages.

When choosing the kernel version for an industrial IoT device, we considered the following factors:

-. Stable. This is the key factor for an industrial application.
-. Long Term Maintenance. Industrial devices tend to have long lifecycles; the longer the maintenance time, the better.

This means we should at least choose the `Longterm` release category. This is why we chose `6.1`, which was the latest LTS version when we were planning the latest kernel version bump.

Another factor is that Debian 12 also uses 6.1 as its kernel version, so we are aligned with Debian 12.

Additionally, the CIP [2] project provides a super-long-term stable kernel program to extend the maintenance lifecycle for a minimum of 10 years after its initial release. This is why we chose the CIP kernel flavor (the `-cipN` postfix).

Regarding the CVE-2024-1086 patch, from [3], you can see it has been backported into v6.1 LTS, specifically the v6.1.76 version. So it will be integrated into the example image kernel in the next kernel version bump.

1. https://www.kernel.org/category/releases.html
2. https://www.linuxfoundation.org/press/civil-infrastructure-platform-expands-slt-stable-kernel-program
3. [CVE 2024-1086](https://nvd.nist.gov/vuln/detail/CVE-2024-1086)

SCordibella, 2024-08-08:
Thank you @BaochengSu for the clear explanation. 
Do you have any schedule for the next kernel bump?

Best regards,
Stefano.

jan-kiszka, 2024-08-08:
Nothing official yet, but we trying to provide an only minimally patched (if not even patch-free) kernel with the next LTS release (6.12?). But it's not only our queue that still needs a few more flushes, see https://github.com/siemens/linux/commits/jan/iot2050 (to be bumped to 6.11-rc X soon).

jan-kiszka, 2024-08-08:
Oh, and note that https://github.com/siemens/linux/commits/jan/iot2050 is not 100% feature-complete /wrt the current 6.1 kernel. We are missing some to-be-discussed patches from the TI SDK in upstream, but also a few more cosmetic changes from our own queue.

SCordibella, 2024-09-11:
Thanks @jan-kiszka for the information.

closed 2024-09-11 by SCordibella