siemens/meta-iot2050 #579: Reboot problem still persists

Int3rC3pt0r, 2025-01-30:
Hello,

Since #440 should be solved I compiled a new swupdate image based on the v01.04.04 release.
When installing the swupdate on an IOT that ran an older image (without the u-boot patches of #570) with an uptime of 140 days the reboot still didn't work. The status LED on the IOT was red.
After a hard resetting the IOT by taking away the power, the device booted into the updated partition without issues.
Is there a way to prevent this from happening?

Thanks in advance.

jan-kiszka, 2025-01-30:
Did you actually update the firmware as well, not just the disk image? Firmware is not managed via swupdate because it requires "special care" (no fallback boot available).

simonabler, 2025-01-30:
How can I install the firmware? I assume, I can do this with `iot2050-firmware-update **.tar.xz`. 
Where do I get the tar file from? is it available in the kas build folder?

SCordibella, 2025-01-30:
@simonabler the firmware is not available in OS project.
If you want to deploy it you need to download and add it to your build.
I extend the `iot2050-firmware-update` recipe to include it in this way:

```
#
# Add firmware update package to the image.
# The update command will be: 
# iot2050-firmware-update /usr/share/iot2050/IOT2050-FW-Update-PKG-V01.04.04-0-gd3b7b08.tar.xz
#
# Author:
#  Stefano Cordibella <stefano.cordibella@edalab.it>
#

FILESEXTRAPATHS:prepend := "${THISDIR}/files:"

SRC_URI:append = " file://IOT2050-FW-Update-PKG-V01.04.04-0-gd3b7b08.tar.xz;unpack=0 "

do_install:append() {
    install -v -d ${D}/usr/share/iot2050/
    install -v -m 644 ${WORKDIR}/IOT2050-FW-Update-PKG-V01.04.04-0-gd3b7b08.tar.xz ${D}/usr/share/iot2050/
}
```

Of course you need to run the update manually on the iot2050 image running.

SCordibella, 2025-01-30:
I forgot to say that the package `u-boot-env` is required. You must add it to your build or better add it as dependency of the `iot2050-firmware-update` recipe.

SCordibella, 2025-01-30:
I forgot to say that the package `u-boot-tools` is required. You must add it to your build or better add it as dependency of the `iot2050-firmware-update` recipe.

simonabler, 2025-01-30:
Worked! 
but i had intall the u-boot-tools before the firmware-update

thanks