siemens/meta-iot2050 #580: SWUpdate Signing workflow

SCordibella, 2025-02-03:
Dear All,
    I see the recipes `swupdate-certificates`and `swupdate-certificates-key`.
I read [here](https://gitlab.com/cip-project/cip-core/isar-cip-core/-/blob/master/doc/README.swupdate.md?ref_type=heads#swupdate-signing) they are created to implement the swupdate signing. In fact the sw-descriptor.sig is already present in the generated swu. First question: what is the command used to sign the image?

Supposing that I want to implement a custom sign procedure I need to change the sign command, and as far as I understand the cip-core layer with the swupdate-signer feature is not already available for iot2050, is it correct?
Are there any plan to integrate it? How can I change the sign command right now?

To check the signed image before installing it I guess that I need to change the swupdate command line on the iot 2050 adding the `-k` flag (supposing that I am using CMS). Are there any examples of this? Is the actual signed image not verified on target?

Best regards,
Stefano.

SCordibella, 2025-02-13:
Hi all, can any of @jan-kiszka, @huaqianli, or @BaochengSu help me with this?"

jan-kiszka, 2025-02-14:
Unfortunately, meta-iot2050 still uses an older version of isar-cip-core. Briefly after the revision references here, upstream got https://gitlab.com/cip-project/cip-core/isar-cip-core/-/commit/980cd0089ed0b0318977522882c4f9d520f6688e which better supports custom signing, including the case of hardware security modules.

The command to sign is now found in https://gitlab.com/cip-project/cip-core/isar-cip-core/-/blob/next/recipes-devtools/swupdate-signer/files/sign-swu-cms, before that, it was in https://gitlab.com/cip-project/cip-core/isar-cip-core/-/blob/22346c57c9461da765151465d492f675d6e4efad/recipes-devtools/swupdate-certificates/files/sign-swu-rsa.

But also the version of isar-cip-core used here enforces swu signing and automatic verification when trying to apply an swu. In fact, the swupdate package refuses to take unsigned updates. It's "just" that the demo signature used here is not providing security as it uses a publicly known key.

SCordibella, 2025-02-24:
Hi @jan-kiszka thank you for the information.
Do you have any plan to update isar-cip-core in meta-iot2050?

SCordibella, 2025-02-24:
Hi @jan-kiszka, I've figured out what needs to be changed in the signing process, and I think it's necessary to modify the verification part as well. If I switch the signing script from `openssl cms` to `openssl dgst`, will I need to adjust anything on the verification side?

SCordibella, 2025-03-06:
Just as update I want to use the pkcs11 standard, but when I try to install the required packages in my image I have the following error:
```
| E: Failed to fetch http://deb.debian.org/debian/pool/main/o/opensc/opensc-pkcs11_0.23.0-0.3%2bdeb12u1_arm64.deb  404  Not Found [IP: 151.101.2.132 80]
| E: Failed to fetch http://deb.debian.org/debian/pool/main/o/opensc/opensc_0.23.0-0.3%2bdeb12u1_arm64.deb  404  Not Found [IP: 151.101.2.132 80]
```
On the debian site there is an updated version of the package 0.23.0-0.3+deb12u2.
Even if I select the wanted version with opensc=0.23.0-0.3+deb12u2 I am unable to fetch the right version.
How can I update the reference to the new package?

jan-kiszka, 2025-03-11:
Looks like you are failing a partial rebuild. Retry after cleaning everything so that the local package database points to the latest version of opensc.

SCordibella, 2025-03-11:
Yes it works after a `cleanall` . Now I am able to load the required library to sign the swu in the imager via the `IMAGER_INSTALL` variable.

Now I don't know how to initialize the environment with my private keys. I try to share the a softhsm2 configuration and token folder with the container and it works, but bitbake can't access to it (I mean my sign script can't list the conf file or access the token folder).

I share the configuration and the token folder via   `--runtime-args "-v $HOME/.config/softhsm2:/tmp/softhsm2"` passed to `kas-container` and then I put this lines in the script for testing:
```
export SOFTHSM2_CONF=/tmp/softhsm2/kas-softhsm2.conf 
echo $SOFTHSM2_CONF
ls -l $SOFTHSM2_CONF
cat $SOFTHSM2_CONF
ls -l /tmp/softhsm2
softhsm2-util --show-slots
```

The output is
```
| /tmp/softhsm2/kas-softhsm2.conf
| ls: cannot access '/tmp/softhsm2/kas-softhsm2.conf': No such file or directory
| cat: /tmp/softhsm2/kas-softhsm2.conf: No such file or directory
| ls: cannot access '/tmp/softhsm2': No such file or directory
| ERROR: Could not load the SoftHSM configuration.
| ERROR: Please verify that the SoftHSM configuration is correct.
```

How can I grant access (or create) a vault with my private keys in the build environment? Is there a way to link my build machine's environment to the container to grant access to the vault?

SCordibella, 2025-03-14:
I managed to handle the issue by adding the script and the vault for SoftHSM2 to the recipe. Now, based on the log, it seems that the image is not being verified on the IoT2050. 
```
root@iot2050-debian:~# swupdate -v -i /media/iot2050-image-swu-example-iot2050-debian-iot2050.swu                                                                                                                  
[DEBUG] : SWUPDATE running :  [read_settings_file] : Reading config file /etc/swupdate.cfg
[DEBUG] : SWUPDATE running :  [read_module_settings] : Reading config settings for module globals
[DEBUG] : SWUPDATE running :  [read_module_settings] : No config settings found for module logcolors
[DEBUG] : SWUPDATE running :  [read_module_settings] : No config settings found for module processes
[TRACE] : SWUPDATE running :  [swupdate_dgst_init] : public key / cert /usr/share/swupdate-signing/swupdate-sign.crt ignored, you need to set SIGALG
```
I thought the verification was enabled.
I thought that if the verification process was not enabled, the image update would be rejected.
Is there a reference example for enabling image verification on iot2050?

SCordibella, 2025-04-02:
Finally I was able to add image verification on iot2050, I extend the swupdate recipe and define `SWU_SIGNED = "1" `. This seems very strange to me, but I noticed that the `SWU_SIGNED` variable is defined only in the class for the swupdate image and not in the recipe...

At the moment, I'm experiencing an issue with the startup of swupdate. The process starts correctly at boot, but after some time (between 30 and 90 seconds), it terminates. It appears to receive a kill signal from one of its child processes, most likely Mongoose. I have observed that this behavior consistently occurs when the system is started without an Ethernet connection, although in some cases the process may still be terminated even when the cable is connected. If I manually restart swupdate afterwards (via systemctl), the process starts and remains active without any issues. Could there be a correlation between the suspected Mongoose crash and the network interface status? I’ve read that a more recent version of swupdate addresses issues of this kind — is there any plan to update swupdate to version 2024.12.1? I am attaching the log from one of my tests and the custom configuration file (that I put in `/etc/swupdate/conf.d/`).

[log.txt](https://github.com/user-attachments/files/19562578/log.txt)

[swupdate-customizations.txt](https://github.com/user-attachments/files/19562582/swupdate-customizations.txt)

closed 2025-04-02 by SCordibella