siemens/meta-iot2050 #583: System Freeze Issue on IOT2050 After OS Update

VikyFlow, 2025-02-12:
Dear all,

I am experiencing an issue with my application running on an IOT2050 after upgrading the operating system from Debian 6.1.54-cip6 to Debian 6.1.102-cip26. 

My current configuration is:
_**Operating System: Debian 6.1.102-cip26
Firmware: IOT2050-FW-Update-PKG-V01.04.04-0**_

Following this update, the system occasionally freezes and stops reading the network interface, even though the interface is still detected by the `nmcli show con` command. 

This issue did not occur with the previous OS version.

Have you encountered similar issues with this OS version? Are there any known solutions or updates available to resolve this problem?

I look forward to your response.

Best regards,
Victoria

huaqianli, 2025-02-12:
Could you share the log and your device info?

Thanks.

VikyFlow, 2025-02-12:
The error occured again even if I was connected thru serial,
I pinged my internal interface and it responds 👍🏻 
i tried to remote access it doents work👎🏻 

My modules dont work. 
Tried to reconnect thru serial, it is super slow .. trying commands sometimes i get outputs sometime no.
And then after I sent a command  i goot my system rebooted:
root@iot2050-debian:~# systemctl status daemon
Failed to get properties: Connection timed out
root@iot2050-debian:~# AVS@[1100 1150 1150]
SIMATIC IOT2050 SE-Boot Version: V01.04.01-0-g629b172b-0x0000
BuildDate: 20240103
SYSFW ABI: 3.1 [version: 21] [21.9.1--v2021.09a (Terrific Lla]

jan-kiszka, 2025-02-13:
Can you still get the kernel messages (`dmesg`) when the situation happened? A complete view around that event would be good.

Is `ip link` reporting the link on the problematic interface as still present? Are you still receiving packets on that line? Does the remote side see your pings, you just don't get its replies?

VikyFlow, 2025-02-13:
HI,
this are the logs when the system was off:
***@***.***:~# dmesg
[    0.000000] Booting Linux on physical CPU 0x0000000000 [0x410fd034]
[    0.000000] Linux version 6.1.102-cip26 ***@***.***)
(aarch64-linux-gnu-gcc (Debian 12.2.0-14) 12.2.0, GNU ld (GNU Binutils for
Debian) 2.40) #1 SMP PREEMPT Thu, 01 Jan 1970 01:00:00 +0000
[    0.000000] Machine model: SIMATIC IOT2050 Basic PG2
[    0.000000] earlycon: ns16550a0 at MMIO32 0x0000000002810000 (options '')
[    0.000000] printk: bootconsole [ns16550a0] enabled
[    0.000000] efi: EFI v2.100 by Das U-Boot
[    0.000000] efi: RTPROP=0xbcec7040 SMBIOS=0xbcec6000 INITRD=0xb4b6b040
MEMRESERVE=0xb4b6a040
[    0.000000] Reserved memory: created DMA memory pool at
0x00000000a0000000, size 1 MiB
[    0.000000] OF: reserved mem: initialized node ***@***.***,
compatible id shared-dma-pool
[    0.000000] Reserved memory: created DMA memory pool at
0x00000000a0100000, size 15 MiB
[    0.000000] OF: reserved mem: initialized node ***@***.***,
compatible id shared-dma-pool
[    0.000000] Reserved memory: created DMA memory pool at
0x00000000a1000000, size 1 MiB
[    0.000000] OF: reserved mem: initialized node ***@***.***,
compatible id shared-dma-pool
[    0.000000] Reserved memory: created DMA memory pool at
0x00000000a1100000, size 15 MiB
[    0.000000] OF: reserved mem: initialized node ***@***.***,
compatible id shared-dma-pool
[    0.000000] software IO TLB: Reserved memory: created restricted DMA
pool at 0x00000000ba000000, size 64 MiB
[    0.000000] OF: reserved mem: initialized node ***@***.***,
compatible id restricted-dma-pool
[    0.000000] Zone ranges:
[    0.000000]   DMA      [mem 0x0000000080000000-0x00000000bfffffff]
[    0.000000]   DMA32    empty
[    0.000000]   Normal   empty
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000080000000-0x000000009e7fffff]
[    0.000000]   node   0: [mem 0x000000009e800000-0x00000000a2200fff]
[    0.000000]   node   0: [mem 0x00000000a2201000-0x00000000bff1ffff]
[    0.000000]   node   0: [mem 0x00000000bff20000-0x00000000bff2ffff]
[    0.000000]   node   0: [mem 0x00000000bff30000-0x00000000bfffffff]
[    0.000000] Initmem setup node 0 [mem
0x0000000080000000-0x00000000bfffffff]
[    0.000000] cma: Reserved 24 MiB at 0x00000000b8800000
[    0.000000] psci: probing for conduit method from DT.
[    0.000000] psci: PSCIv1.1 detected in firmware.
[    0.000000] psci: Using standard PSCI v0.2 function IDs
[    0.000000] psci: Trusted OS migration not required
[    0.000000] psci: SMC Calling Convention v1.2
[    0.000000] percpu: Embedded 19 pages/cpu s40360 r8192 d29272 u77824
[    0.000000] pcpu-alloc: s40360 r8192 d29272 u77824 alloc=19*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1
[    0.000000] Detected VIPT I-cache on CPU0
[    0.000000] CPU features: detected: GIC system register CPU interface
[    0.000000] CPU features: detected: ARM erratum 845719
[    0.000000] alternatives: applying boot alternatives
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 258048
[    0.000000] Kernel command line: console=ttyS3,115200n8
earlycon=ns16550a,mmio32,0x02810000 rootwait watchdog.handle_boot_enabled=0
usbcore.nousb
[    0.000000] Dentry cache hash table entries: 131072 (order: 8, 1048576
bytes, linear)
[    0.000000] Inode-cache hash table entries: 65536 (order: 7, 524288
bytes, linear)
[    0.000000] mem auto-init: stack:all(zero), heap alloc:off, heap free:off
[    0.000000] Memory: 850936K/1048576K available (12032K kernel code,
1090K rwdata, 4032K rodata, 1984K init, 432K bss, 173064K reserved, 24576K
[... 605 lines cut ...]
[   16.262455] pps pps0: new PPS source ptp0
[   16.345878] remoteproc remoteproc0: powering up b034000.pru
[   16.392654] remoteproc remoteproc0: Booting fw image
ti-pruss/am65x-sr2-pru0-prueth-fw.elf, size 38800
[   16.406207] remoteproc remoteproc0: unsupported resource 5
[   16.418471] remoteproc remoteproc0: remote processor b034000.pru is now
up
[   16.428379] remoteproc remoteproc1: powering up b004000.rtu
[   16.470175] remoteproc remoteproc1: Booting fw image
ti-pruss/am65x-sr2-rtu0-prueth-fw.elf, size 30968
[   16.479767] remoteproc remoteproc1: remote processor b004000.rtu is now
up
[   16.486801] remoteproc remoteproc2: powering up b00a000.txpru
[   16.501273] remoteproc remoteproc2: Booting fw image
ti-pruss/am65x-sr2-txpru0-prueth-fw.elf, size 37328
[   16.512122] remoteproc remoteproc2: remote processor b00a000.txpru is
now up
[   18.587567] icssg-prueth icssg0-eth eno2: Link is Up - 100Mbps/Full -
flow control off
[   18.595794] IPv6: ADDRCONF(NETDEV_CHANGE): eno2: link becomes ready
[   20.379395] icssg-prueth icssg0-eth eno1: Link is Up - 1Gbps/Full - flow
control off
[   20.387256] IPv6: ADDRCONF(NETDEV_CHANGE): eno1: link becomes ready

Il giorno gio 13 feb 2025 alle ore 06:35 Jan Kiszka <

jan-kiszka, 2025-02-13:
Ok, if that dump contains the time when the network connection started to fail, then the kernel does not notice this in any way.

How about my other questions?

VikyFlow, 2025-02-14:
Hi.

this is my results when my app isnt working:

***@***.***:~# ip link
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN mode
DEFAULT group default qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
2: eno2: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc mq state UP mode
DEFAULT group default qlen 1000
    link/ether 4c:e7:05:85:af:08 brd ff:ff:ff:ff:ff:ff
3: eno1: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc mq state UP mode
DEFAULT group default qlen 1000
    link/ether 4c:e7:05:85:af:07 brd ff:ff:ff:ff:ff:ff

it happened and is always connected :
Every 1.0s: nmcli device status

 iot2050-debian: Fri Feb 14 09:31:53 2025

DEVICE  TYPE      STATE                   CONNECTION
eno1    ethernet  connected               eno1-default
eno2    ethernet  connected               Wired connection 1
lo      loopback  connected (externally)  lo

when i ping the interface i receive answers ..
it seems like in my sysstem the connection is lost and then alll my
services brake down..

jan-kiszka, 2025-02-14:
OK, if pinging works, then it is likely not a networking issue of the OS, rather an application problem.

Can you still reach the 2050 via ssh? Try to narrow down what is still working and what not. In some scenarios (not known for this device, though), only small packets still make it while larger get stuck somewhere. You can test that by increasing the packet size in your pings e.g.

VikyFlow, 2025-02-14:
Hi sorry,
Its most likely a problem of the hardware.

We did so many tests.

NO I cannot reach the IOT thru ssh when the system is down.

jan-kiszka, 2025-02-14:
As I wrote above: narrow down the issue, please. The kernel sees no problem, small-size pings still work, but ssh and other app traffic fail - we are missing some piece in that puzzle in between.

VikyFlow, 2025-02-17:
HI,
How can we narrow down the issue?
I want to emphasise that both **SSH** and the **serial** connection are not working while my system doesnt work.
I can connect via serial, but it's almost as if I'm not connected because it doesn't respond to commands and is extremely slow, even when typing. However, ping still works on both interfaces.

jan-kiszka, 2025-02-17:
That is a state we haven't seen yet (to my best knowledge).

Is the system under heavy load (`top`)?

Do interrupts still arrive (`/proc/interrupts`)? Specifically, are timer events still coming?

How reliably can you reproduce the state? Every variable you can remove from the reproduction pattern may be helpful for eventually having the same on our side.

VikyFlow, 2025-02-18:
Hi,
when the error happens i found out this logs related to the network
interfaces; maybe can be helpful to understand better:

***@***.***:~# journalctl -u NetworkManager --since "1 hour ago" |
grep -i "disconnect\|down\|failed"
Feb 18 10:23:45 iot2050-debian NetworkManager[420]: <info>
 [1739874225.6418] monitoring ifupdown state file '/run/network/ifstate'.
Feb 18 10:23:46 iot2050-debian NetworkManager[420]: <info>
 [1739874226.6602] settings: Loaded settings plugin: ifupdown
("/usr/lib/aarch64-linux-gnu/NetworkManager/1.42.4/libnm-settings-plugin-ifupdown.so")
Feb 18 10:23:46 iot2050-debian NetworkManager[420]: <info>
 [1739874226.6606] ifupdown: management mode: unmanaged
Feb 18 10:23:46 iot2050-debian NetworkManager[420]: <info>
 [1739874226.6608] ifupdown:       interface-parser: parsing file
/etc/network/interfaces
Feb 18 10:23:46 iot2050-debian NetworkManager[420]: <info>
 [1739874226.6609] ifupdown:       interface-parser: source line includes
interfaces file(s) /etc/network/interfaces.d/*
Feb 18 10:23:46 iot2050-debian NetworkManager[420]: <info>
 [1739874226.6611] ifupdown: interfaces file /etc/network/interfaces.d/*
doesn't exist
Feb 18 10:23:46 iot2050-debian NetworkManager[420]: <info>
 [1739874226.6612] ifupdown:       interface-parser: finished parsing file
/etc/network/interfaces
Feb 18 10:23:46 iot2050-debian NetworkManager[420]: <info>
 [1739874226.8184] device (lo): state change: unavailable -> disconnected
(reason 'connection-assumed', sys-iface-state: 'external')
Feb 18 10:23:47 iot2050-debian NetworkManager[420]: <info>
 [1739874227.4837] device (lo): state change: disconnected -> prepare
(reason 'none', sys-iface-state: 'external')
Feb 18 10:23:49 iot2050-debian NetworkManager[420]: <info>
 [1739874229.4058] device (eno2): state change: unavailable -> disconnected
(reason 'carrier-changed', sys-iface-state: 'managed')
Feb 18 10:23:49 iot2050-debian NetworkManager[420]: <info>
 [1739874229.4345] device (eno2): state change: disconnected -> prepare
(reason 'none', sys-iface-state: 'managed')
Feb 18 10:23:51 iot2050-debian NetworkManager[420]: <info>
 [1739874231.2656] device (eno1): state change: unavailable -> disconnected
(reason 'carrier-changed', sys-iface-state: 'managed')
Feb 18 10:23:51 iot2050-debian NetworkManager[420]: <info>
 [1739874231.2869] device (eno1): state change: disconnected -> prepare
(reason 'none', sys-iface-state: 'managed')

Thank you,
Victoria

jan-kiszka, 2025-02-18:
NetworkManager sees a carrier drop at some point. Does it see a re-establishment as well (your filtering might have suppressed messages)?

In any case, `ip link` reported the link up in the error state.

Any insights about link state from the switch or the peer (if cross-linked)?

VikyFlow, 2025-02-21:
Hi,
so I managed to see if maybe if my update runtime version had to do with these problem but actually the same problems reappears but in different log description:

I see this message on the kernel that could be correlated to the issue I'm facing:

`Dec 01 14:28:29 iot2050-debian kernel: keystone-pcie 5600000.pcie: Phy link never came up`

Could this help? How can I investigate more related to this issue?

``