siemens/meta-iot2050 #591: switchserialmode error

RoyBellingan, 2025-03-18:
The code in 
https://github.com/siemens/meta-iot2050/blob/cbbe0f4a02ee74c93f6a67a8222550e630870e15/recipes-app/iot2050setup/files/iot2050setup.py#L518

References what I think are older flag, no longer present 
https://github.com/siemens/meta-iot2050/blob/950a0aaa017e1cd39e4a1eb535b2344325209611/recipes-app/switchserialmode/files/src/switchserialmode.c#L111

Specifically -g and -D are no longer present, also the first parameter (cp210x in the example linked) looks like has no effect.

BaochengSu, 2025-03-19:
@RoyBellingan We've refactored both the implementation of `switchserialmode` and its CLI interface in d1432de5e39a6125bf0f90d5a35477d3ba81e0cb. The cli commands in `iot2050setup.py` were changed accordingly in the same PR #369. I think you are referring to the older version of `iot2050setup.py`.

This change is released in V01.04.01, I think the 1st forum link you are referring is before the V01.04.01.

The second forum link has never mentioned the cli, but the iot2050setup TUI tool. We've kept in mind of the consistence of CLI & TUI during the refactoring, so the method described in this forum post shall not be impacted by this refactoring.

RoyBellingan, 2025-03-19:
@BaochengSu thank you very much !
I was not aware of all those detail

I will now  base my search on https://github.com/siemens/meta-iot2050/blob/master/recipes-app/board-conf-tools/files/board-conf-tools/iot2050setup.py !

closed 2025-03-19 by RoyBellingan