siemens/meta-iot2050 #595: Disable swupdate at boot

SCordibella, 2025-04-08:
Dear Support,

I am trying to achieve a relatively simple goal: disable the swupdate service at boot in a custom image that inherits from the swupdate example image.

The rationale behind this is that the user must configure a specific password on the device before the update infrastructure can be activated. After this initial setup, the service will be manually enabled and started via systemctl enable / start.

However, despite trying different approaches to disable the service, it is still present and running after the system boots. I suspect I may be missing something related to how `deb-systemd-helper` works, especially with its `*.dsh-also` links or other related mechanisms.

As I am not very familiar with Debian's internals, could you kindly advise on the correct and most reliable way to prevent the service from starting at boot?

Thank you in advance for your support.

Best regards,
Stefano

jan-kiszka, 2025-04-10:
Without going into all details: You could define your own customization package (https://github.com/siemens/meta-iot2050/tree/master/recipes-core/customizations-example) that `DEBIAN_DEPENDS` on swupdate to ensure that is installed first and then runs `systemd disable swupdate` in its postinst maintainer script.

SCordibella, 2025-04-11:
Thank you @jan-kiszka for the suggestion, I will try as soon as possible.

closed 2025-05-05 by SCordibella