siemens/meta-iot2050 #614: upstream build failed with fetch failures

arulcse2008, 2025-08-05:
Build failed due to fetch failures for ti-pruss-firmware package, Seems like https://git.ti.com/cgit/processor-firmware/ti-linux-firmware/ has been updated recently. 

Parsing recipes: 100% |#########################################################################| Time: 0:00:00
Parsing of 124 .bb files complete (115 cached, 9 parsed). 370 targets, 0 skipped, 1 masked, 0 errors.
NOTE: Resolving any missing task queue dependencies
Sstate summary: Wanted 57 Local 0 Mirrors 0 Missed 57 Current 0 (0% match, 0% complete)#        | ETA:  0:00:00
Initialising tasks: 100% |######################################################################| Time: 0:00:00
NOTE: Executing Tasks
WARNING: ti-pruss-firmware-09.00.00.005-r0 do_fetch: Failed to fetch URL https://git.ti.com/cgit/processor-firmware/ti-linux-firmware/plain/ti-pruss/am65x-pru1-prueth-fw.elf?h=09.00.00.005;downloadfilename=am65x-pru1-prueth-fw.elf;sha256sum=56cb71d205810a3609a5dff950fa7d7ea6b383d9d2fb6b1f2116ba58527f2f7f, attempting MIRRORS if available
ERROR: ti-pruss-firmware-09.00.00.005-r0 do_fetch: Fetcher failure: Fetch command export PSEUDO_DISABLED=1; export GIT_PROXY_COMMAND="oe-git-proxy"; export NO_PROXY="*"; export PATH="/work/isar/scripts:/work/isar/bitbake/bin:/usr/sbin:/usr/bin:/sbin:/bin"; export HOME="/tmp/tmprmiw0n5m"; /usr/bin/env wget -t 2 -T 30 -O /build/downloads/am65x-pru1-prueth-fw.elf.tmp -P /build/downloads 'https://git.ti.com/cgit/processor-firmware/ti-linux-firmware/plain/ti-pruss/am65x-pru1-prueth-fw.elf?h=09.00.00.005' --progress=dot -v failed with exit code 4, see logfile for output
ERROR: ti-pruss-firmware-09.00.00.005-r0 do_fetch: Fetcher failure for URL: 'https://git.ti.com/cgit/processor-firmware/ti-linux-firmware/plain/ti-pruss/am65x-pru1-prueth-fw.elf?h=09.00.00.005;downloadfilename=am65x-pru1-prueth-fw.elf;sha256sum=56cb71d205810a3609a5dff950fa7d7ea6b383d9d2fb6b1f2116ba58527f2f7f'. Unable to fetch URL from any source.
ERROR: Logfile of failure stored in: /build/tmp/work/iot2050-debian-arm64/ti-pruss-firmware/09.00.00.005-r0/temp/log.do_fetch.304
ERROR: Task (/build/../repo/meta/recipes-bsp/ti-pruss-firmware/ti-pruss-firmware_09.00.00.005.bb:do_fetch) failed with exit code '1'
NOTE: Tasks Summary: Attempted 425 tasks of which 410 didn't need to be rerun and 1 failed.

Summary: 1 task failed:
  /build/../repo/meta/recipes-bsp/ti-pruss-firmware/ti-pruss-firmware_09.00.00.005.bb:do_fetch
Summary: There was 1 WARNING message.
Summary: There were 2 ERROR messages, returning a non-zero exit code.
2025-08-05 05:51:52 - ERROR    - Command "/work/isar/bitbake/bin/bitbake -c build iot2050-image-swu-example" failed with error 1
bhavani@bhavani:~/arul/oe/industrial-distro/meta-iot2050$ (master)

huaqianli, 2025-08-06:
> Hi [@arulcse2008](https://github.com/arulcse2008) ,
> 
> The error is occurring because git.ti.com is unstable. To work around this issue, I recommend backing up your downloads directory to avoid having to re-download files.
> <img alt="Image" width="851" height="430" src="https://private-user-images.githubusercontent.com/128661405/474776508-49759f9c-2b4e-4959-9bc1-42124f3a275a.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NTQ0Njg5ODIsIm5iZiI6MTc1NDQ2ODY4MiwicGF0aCI6Ii8xMjg2NjE0MDUvNDc0Nzc2NTA4LTQ5NzU5ZjljLTJiNGUtNDk1OS05YmMxLTQyMTI0ZjNhMjc1YS5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjUwODA2JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI1MDgwNlQwODI0NDJaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT02YmZiYzVkOTA5MjM2ZDQ0YmM5Mjc5M2ZhYzE1MDljOWI2ODYzYTk3ZjAwYzJhOTA2YzQyN2M2OWEyYzk3ZDYxJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.pwfCX9gM3MFZ4kYzVtbFiyq5SElYVMlZhHJJe6Pf27g">
> 
> Thanks.

Since this is a third-party server reliability issue rather than a bug in our codebase, I'm closing this PR. Feel free to reopen if you discover the issue is actually related to our build system.

Thanks for your understanding.

jan-kiszka, 2025-08-20:
TI addressed the issue via mirror: https://github.com/TexasInstruments/ti-linux-firmware. We should switch to it for better reliability.

closed 2025-08-22 by huaqianli